Alpha-LiDAR/alpha_lidar #1: test dataset

chengwei0427, 2024-09-20:
Hi, @HViktorTsoi 

Great work! Are there any publicly available datasets for SLAM testing?

HViktorTsoi, 2024-09-20:
Yes！We are uploading the datasets in the next few days

chengwei0427, 2024-09-24:
Thanks for your work, and looking forward to the dataset being open-sourced.

chengwei0427, 2024-09-29:
Hi, @HViktorTsoi @yuzehh 
I downloaded the data you shared in your alpha_lidar repository. However, since the LiDAR keeps rotating from the moment the bag is opened, I have tested FastLIO2, Point-LIO, and IG-LIO, but none of them are running correctly. Could you please advise on how to make this data work? Is dynamic initialization necessary to get it running?

HViktorTsoi, 2024-09-29:
Hi, the datasets may be challenging for other LIOs to initialize due to rapid rotation from the begining. αLiDAR has a  non-static initialization strategy when starting state estimation. Please test with our uploaded code and see the initialization part. 

Also please check if the [initial LiDAR-IMU extrinsics](https://github.com/HViktorTsoi/alpha_lidar/blob/a336d02454decaba03336e24fa650bfa089e1f69/software/alpha_lidar_ws/src/state_estimation/config/rs16_rotation_v2.yaml#L30) are correct.

satyajitghana, 2024-10-03:
@chengwei0427 i think you'll also need to fuse encoder data as well to state estimation in other slam algorithms

https://github.com/HViktorTsoi/alpha_lidar/blob/4f7c6ef2f747e8992846cadd01952fbbaaa8ce0f/software/alpha_lidar_ws/src/state_estimation/src/voxelMapping.cpp#L1196-L1208

closed 2024-11-26 by HViktorTsoi