Alpha-LiDAR/alpha_lidar #13: Regarding the source of the topic /imu/encoder

Luthermorning, 2024-11-19:
Thank you for your open source work for contributing to the community, I would like to know how your encoder data is emitted from STM32 and received in ROS, can you discuss this with you.

HViktorTsoi, 2024-11-26:
Hi,  the data link path of the encoder message is like this:

Motor --> [Encoder angle packets] --> CAN --> stm32 --> [Timestamping and packing the angle packets together with adjecent IMU packets] --> USART --> Host machine --> ROS serial driver --> [Decode both IMU and angle packets and publish them as /imu/data and /imu/encoder]

Luthermorning, 2024-11-27:
> Hi, the data link path of the encoder message is like this:
>
> Motor --> [Encoder angle packets] --> CAN --> stm32 --> [Timestamping and packing the angle packets together with adjecent IMU packets] [...]

Thanks, it`s cool.

closed 2024-11-29 by HViktorTsoi