Alpha-LiDAR/alpha_lidar #15: Slip Ring and 24V Battery

85256638, 2024-11-24:
[image]

Hi, could you kindly provide details about the slip ring and the 24V battery you used in this project?
Thanks!

HViktorTsoi, 2024-11-26:
Hi, we use a Senring slip ring (12 line).  The 24V battery can be any off-the-shelf Lithum-ion battery with at least 1A current support, e.g., this one [Taobao Link](https://item.taobao.com/item.htm?abbucket=17&id=660918833068&ns=1&pisk=fPCKp_mnDxYQ3W3tBYwMq7KeZMzG2_QUxM7jZ3xnFGIOPhcu8B2ywgKO4HvHOHcJwaIPqgQz8Ut5VgKlKRVc8wRyNox8iSbEPt92aMHIdFaWkUYQ952gRwRyNu3iN-VA8i1vIEtBPFa9rUnIVgGI1VtwyYGCVH9sfEL2NQO5Vft6zEtSVQtS5PLvo3gBdbg6fUTSF0sWNPQsL-rJf2towTxXepSz5lGoZ1L_Yn_KEbYO_eppDw1KZb4yJpKfRhZD2Pw6HM9RarhOX_KJEnJEl-O1uOjBWIFidI_OFiLFZzckq67CFLCuj7APbTsWyGNqqBWcYsJO7-G17BKwht7TaXxBn6b2Wdo_4FB1ug8XpcDF4BL54K5u0X-CPZ5VnIFidI_OFiLFZzckq67CFLCujSSyLsfvxBoD1HkQWPHrUpTwJMPK0Z8c3489myF-UYJa7FK0WoDrUp9BWn4LcYkyQif..&priceTId=2150441717326360535271115e3e47&skuId=4764809500857&spm=a21n57.1.item.4.4e88523cuXgHbo&utparam=%7B%22aplus_abtest%22%3A%22f705a8f38c1d1e9c264bef88c7359e5c%22%7D&xxc=taobaoSearch). We actually mount the battery together with the host machine and connect it through a extended power cable. But you can also mount that inside the base like the picture above.

85256638, 2024-11-29:
Thanks a lot for providing details about the slip ring.

I contacted Senring, but their pre-sales person couldn't not identify the exact model number from the CAD file, and the photos from this project page. Therefore, could you kindly provide the detailed part number of the slip ring used? This would help massively! 👍