Alpha-LiDAR/alpha_lidar #16: A summary of the missing information required to re-produce this work

85256638, 2024-12-03:
Thanks a lot for your dedication to open-source this great piece of work. After thoroughly review the readme, I summarised a list of missing information which required to re-produce this work:

1. Exact model number of the sliping  #15 
[image]
_Senring sales team were unable to work out the model used._

2. BOM of the PCB board.
[image]
 _i.e. the capacitor, crystal oscalliator used in this project etc._

3. Mounting structures for Livox Lidars. #14 

5. Information about the red IMU board used for Hesai and Velodyne Lidar
[image]

6. STM32 source code #9 

8.  what is the purpose of this little red button?
[image]

himhan34, 2024-12-23:
@85256638 maybe i wish this helps you 
i  think the slip ring that the author used is  the pic which is in below, especially the  size is 22mm, 12ch, 5A
[image]

[image]

himhan34, 2024-12-23:
me either are looking for the right control panel bom!
[image]

jxx315, 2024-12-24:
[image]
[image]

The hollow shaft of this motor has an inner diameter of 18mm,    Slip Rings with 22mm OD cannot be adapted.

jxx315, 2024-12-24:
> me either are looking for the right control panel bom! [...]

Slip Rings with 22mm OD cannot be adapted.

himhan34, 2024-12-24:
> Slip Rings with 22mm OD cannot be adapted.

as i understood outer diameter is 22mm

jxx315, 2024-12-25:
> as i understood outer diameter is 22mm

The inner diameter of the motor is 18mm, the outer diameter is 22mm, the outer diameter of the slip ring is 22mm, how to insert it?

himhan34, 2024-12-25:
[image]
[image]
[image]

as you can see in here I think we can insert the slip ring cuz the od of the platform is 22mm

himhan34, 2024-12-25:
i think i had given the wrong information so I erased all, after checking the right information I will comment.